nebhead/PiFire #10: Improve Status on Display

slauzon, 2020-09-20:
Improve status on display to provide more information on current status.  Specifically notifying of startup process begin/end and showing igniter status.

nebhead, 2020-09-20:
Very cool idea.  I like this for safety as well.  I'll look to get this implemented soon.

slauzon, 2020-09-20:
I already started to play around with this a bit.  See below.

```
	def DisplayTemp(self, temp):
		with canvas(self.device) as draw:
			font = ImageFont.truetype("impact.ttf", 42)
			font_header = ImageFont.truetype("impact.ttf", 14)
			text = str(temp)[:5]
			(font_width, font_height) = font.getsize(text)
			try:
				draw.text((0, 0), "IGNITER ACTIVE", font=font_header, fill=255)
				draw.text((128//2 - font_width//2, (64//2 - font_height//2)+8), text, font=font, fill=255)
			except:
				now = str(datetime.datetime.now())
				now = now[0:19] # Truncate the microseconds
				print(str(now) + ' Error displaying temperature.')
```
This seemed to work well, but I like your idea of having an IP address displayed on the top and maybe a small fire logo animated or more simply a dot that flashes while it's on.  What do you think?

nebhead, 2020-09-21:
I'm thinking how I might re-architect this function in this module to support more input in a generic sense.  Today, the Display.DisplayTemp() function only takes a simple temperature value as input, but to do what we want to do, I need to pass it more state data, like the state of the igniter, mode, etc.  So give me some time and I'll come up with something.  I may also try to add icons (perhaps from FontAwesome) that could represent different modes.  On this tiny display it might get cramped - but for future upgrades, it would be nice to have flexibility to add more information.

nebhead, 2020-09-22:
OK, done.  :)  On to the next.

closed 2020-09-22 by nebhead