nebhead/PiFire #9: Re-ignite Procedure

slauzon, 2020-09-20:
Develop formula and function for implementing a re-ignite functionality for when the fire goes out.

nebhead, 2020-09-20:
Sounds like another great idea.  I can make this an option in the settings, so that the user can choose whether to try re-igniting and how many times to re-attempt before erroring out.

slauzon, 2020-09-20:
> Sounds like another great idea. I can make this an option in the settings, so that the user can choose whether to try re-igniting and how many times to re-attempt before erroring out.

Great ideas especially the re-attempts before erroring out.  Good safety consideration.

nebhead, 2020-09-24:
OK, implemented a re-ignite mode and I think we can close this one.  Tested it on the prototype build and it seems to be working, but will monitor this on the real system to see how it behaves.

closed 2020-09-24 by nebhead