nebhead/PiFire #25: GPIO Outputs reversed

gsherwin74, 2021-06-27:
I using the new PCB and a Pi Zero W.  When it boots up I am able to see the web interface and everything seems to be working fine however, the relays are active when they unit should be standing by.  When you tap the PLAY button to start the unit, the relays turn off.  When you hit the STOP button to shut the smoker down, the relays all turn on again.  The output on the GPIO's are all backwards.  What is causing that?

nebhead, 2021-07-09:
FYI... I think it's fixed with the below commit in the development branch.  If you could try this and let me know if it works for you, that would be great.  

https://github.com/nebhead/PiFire/commit/9d8e76e4181377be039ee6197119d56addf2a0b7

Once you've pulled the new files, you'll need to edit the settings.json file and reboot. Replace the globals section with the below:

e.g.

"globals": {
    "grill_name": "",
    "debug_mode": false,
    "page_theme": "light",
    "triggerlevel": "HIGH",
    "units": "F"
  },
Reboot, and you should be back in business.

gsherwin74, 2021-07-09:
Excuse my lack of experience with Linux but how do I get this installed on the Pi. I followed your original instructions step by step.  As soon as I get it updated I'll test it out and report back.

9d8e76e<https://github.com/nebhead/PiFire/commit/9d8e76e4181377be039ee6197119d56addf2a0b7>

nebhead, 2021-07-10:
No problem.  Let's try the following.  Use SSH to log into your pi, then: 

```
pi@pifire:~ $ cd pifire/
pi@pifire:~ $ sudo git checkout development
```
This should switch you to the development branch.  Then pull the latest changes.  

```
pi@pifire:~ $ sudo git pull
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 9 (delta 7), reused 1 (delta 0), pack-reused 0
Unpacking objects: 100% (9/9), done.
From https://github.com/nebhead/pifire
   5733325..9d8e76e  development -> origin/development
   62101ab..957ae3d  gh-pages    -> origin/gh-pages
   d3f62b6..5733325  main        -> origin/main
Updating 5733325..9d8e76e
Fast-forward
 common.py              | 23 ++++++++++++++++++++++-
 control.py             | 25 +++++++++++++++++++------
 grillplat_pifire.py    | 49 +++++++++++++++++++++++++++++--------------------
 grillplat_prototype.py | 47 ++++++++++++++++++++++++++---------------------
 4 files changed, 96 insertions(+), 48 deletions(-)
```

Then look at your settings.json file and update the globals section. 

```
pi@pifire:~ $ sudo nano settings.json
```
Find the globals section and replace it with: 
```
"globals": {"grill_name": "","debug_mode": false,"page_theme": "light","triggerlevel": "HIGH","units": "F"},

```
CTRL-O to save, CTRL-X to exit nano. 

Then reboot the system:
```
pi@pifire:~ $ sudo reboot
```

gsherwin74, 2021-07-10:
Got an error when trying to update...

pi@PiFire:~ $ cd pifire/

pi@PiFire:~/pifire $ sudo git checkout development
M       grillplat_pifire.py
Branch 'development' set up to track remote branch 'development' from
'origin'.
Switched to a new branch 'development'

pi@PiFire:~/pifire $ sudo git pull
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 9 (delta 7), reused 1 (delta 0), pack-reused 0
Unpacking objects: 100% (9/9), done.
From https://github.com/nebhead/pifire
   5733325..9d8e76e  development -> origin/development
   62101ab..957ae3d  gh-pages    -> origin/gh-pages
Updating 5733325..9d8e76e
error: Your local changes to the following files would be overwritten by
merge:
        grillplat_pifire.py
Please commit your changes or stash them before you merge.
Aborting

gsherwin74, 2021-07-10:
Nevermind.  I had started doing the change to grillplat_pidfire so I
undid that with  git reset --hard.
It pulled correctly now.

Settings updated.

Testing now...will report back.

Thanks so much for your help on this.

nebhead, 2021-07-10:
Awesome!  I hope this works, I've been feeling bad about the whole thing.  Thanks for letting me know.

gsherwin74, 2021-07-10:
Successful startup and now getting up to temp on HOLD mode.  Does the PID self adjust and learn as goes?

nebhead, 2021-07-10:
Yep, the PID learns as it goes.  It takes a good amount of time (maybe 30m+) for it to really center in on the temperature. Just make sure you don't open and close the lid as that will drop the temp inside the grill and make it swing more.  Admittedly, I'm not very strong in PID theory and hadn't really tuned it for my grill.  I'm using the settings from @DBorello who I think might have had a smaller Traeger than mine.  However, all the PID settings are in the settings page and can be tuned from there.  Someday I'll play around with it to see if there is something that makes improvements over the current settings.

gsherwin74, 2021-07-10:
Ok, it's been about 30 minutes now and this thing is locked in at 218 +/- 5.  This thing is awesome.

I'm thinking about adding a self closing lid so after start up it closes and goes to hold mode.

nebhead, 2021-07-10:
Woohoo!  So glad it's working for you now.  Would be really cool feature to add an auto-closing door.  Anyway, I'm going to close this issue and will merge the changes into main.  Thanks for filing the issue.

closed 2021-07-10 by nebhead