nebhead/PiFire #41: WARNING: History data is not present in database

weberbox, 2021-10-01:
When there is no history the event log is being spammed with tons of messages from the history page requesting data. I commented out WriteLog for now but there is probably a better way to do this. My initially thought was to just check if the grill is active before requesting history data but that would cause the graph to not show any data once you stop a cycle. 

```
Common.py

def ReadHistory(num_items=0, flushhistory=False):

 ..................

else:
		if cmdsts.exists('control:history'):
			if(num_items > 0) and (len(data_list) < num_items):
				# Get range
				liststart = cmdsts.llen('control:history') - num_items
			else: 
				liststart = 0
			data = cmdsts.lrange('control:history', liststart, -1)
			for index in range(len(data)):
				data_list.append(data[index].split(' ', 6))  # Splits out each of the values into seperate list items 
		else:
			event = 'WARNING: History data is not present in database.'
			WriteLog(event)
```

[image]

nebhead, 2021-10-02:
Looking into it.  Thanks!

nebhead, 2021-10-02:
OK, think it's resolved with this update: https://github.com/nebhead/PiFire/commit/9969a8592c4558b810ab6423918664598ec8d03e

Please check it and let me know. 

Basically just added a dummy entry and created a new history DB when this occurs, so that it shouldn't hit multiple times.

weberbox, 2021-10-04:
Huh I didn't even think to add some data. I looked at it and was making it way more complicated than needed. This seems to have solved the issue from the testing I have done. It stops the history graph from updating when there is no data (used to just scroll with no info but times updated) I think it is better to not scroll since there is no data anyway.

closed 2021-10-04 by weberbox