nebhead/PiFire #51: New install

Chadac77, 2021-11-03:
So have tried to  install on pi zero and I get webui but does not  show temp , will not start, no manual control and screen does not work . Same on pi3b. 
Event on start (warning: issue reading /tmp/current.log) 
Any help would be great.
Thanks

nebhead, 2021-11-04:
Hi @Chadac77, 

Are you using a PCB design for the project?  Which version?  What type of relay board do you have?  Is it high or low triggered?  Are you using the latest source from the development branch or are you using the main branch?  

If you have the supervisor UI running, can you check if both the control script and the web app script are running?  (goto http://your.pifire.ip.addr:9001)

Chadac77, 2021-11-04:
No pcb just bread board for now. Relay or triggered low. Using the main branch. Screen is SSD1306.
Web app running 
Control runs for a few seconds then exits

nebhead, 2021-11-04:
OK, thanks.  Sounds like control.py is crashing on something.  

Can you send the contents of the following files: 

```
~/pifire/settings.json
~/pifire/logs/control.err.log
~/pifire/logs/control.out.log
```

I believe we'll be able to start to drill down on the error from there.

Chadac77, 2021-11-04:
i get (permission denied)

Chadac77, 2021-11-04:
settings

 {"history_page": {"minutes": 1, "clearhistoryonstart": true, "autorefresh": "on", "datapoints": 60}, "probe_settings": {"probe_profiles": {"TWPS00": {"Vs": 3.28, "Rd": 10000, "A": 0.00073431401, "B": 0.0002157437, "C": 9.515686e-08, "name": "Thermoworks-Pro-Series-HeaterMeter"}, "ET73-HM": {"Vs": 3.28, "Rd": 10000, "A": 0.00024723753, "B": 0.00023402251, "C": 1.3879768e-07, "name": "ET-73-Heatermeter"}, "iGrill-HM": {"Vs": 3.28, "Rd": 10000, "A": 0.0007739251279, "B": 0.0002088025997, "C": 1.154400438e-07, "name": "iGrill-Heatermeter"}, "PT1000-00": {"Vs": 3.37, "Rd": 10000, "A": 0.04136906456, "B": -0.00677987613, "C": 2.760294589e-05, "name": "PT-1000-OEM-RTD"}, "ET73-SP": {"Vs": 3.28, "Rd": 10000, "A": 0.00023067434, "B": 0.00023696596, "C": 1.2636414e-07, "name": "ET-73-skyeperry1"}}, "probes_enabled": [1, 1, 1]}, "globals": {"grill_name": "", "debug_mode": false, "page_theme": "light", "triggerlevel": "LOW", "buttonslevel": "HIGH", "shutdown_timer": 60, "units": "F"}, "ifttt": {"APIKey": ""}, "pushbullet": {"APIKey": "", "PublicURL": ""}, "pushover": {"APIKey": "", "UserKeys": "", "PublicURL": ""}, "probe_types": {"grill0type": "PT1000-00", "probe1type": "TWPS00", "probe2type": "TWPS00"}, "outpins": {"power": 4, "auger": 14, "fan": 15, "igniter": 18}, "inpins": {"selector": 17}, "cycle_data": {"PB": 60.0, "Ti": 180.0, "Td": 45.0, "HoldCycleTime": 20, "SmokeCycleTime": 15, "PMode": 2, "u_min": 0.15, "u_max": 1.0}, "smoke_plus": {"enabled": false, "min_temp": 160, "max_temp": 220, "cycle": 10, "frequency": 1, "duty_cycle": 50}, "safety": {"minstartuptemp": 75, "maxstartuptemp": 100, "maxtemp": 550, "reigniteretries": 1}, "modules": {"grillplat": "pifire", "adc": "ads1115", "display": "ssd1306", "dist": "prototype"}}

Chadac77, 2021-11-04:
had to use filezilla
this is my 
~/pifire/logs/control.err.log
~/pifire/logs/control.out.log

https://drive.google.com/drive/folders/1BJVltCiJdCBBAnAsJ1JZl26SeANqM6oO?usp=sharing

nebhead, 2021-11-04:
Thanks for providing the files.  Can you approve my request for access to the drive link?  I just submitted the request (nebhead77 @ gmail).

Chadac77, 2021-11-04:
done

nebhead, 2021-11-04:
> done

Hmm... still don't seem to have access.  Could you make it public?

Chadac77, 2021-11-04:
try this
https://drive.google.com/drive/folders/1BJVltCiJdCBBAnAsJ1JZl26SeANqM6oO?usp=sharing

nebhead, 2021-11-04:
> done

OK, sorry, got access, not sure why it took a while to work. :) 

Anyway, seems we have an issue with the display.  It appears that the display might not be recognized.  Can you make sure it's hooked up right?  

```
luma.core.error.DeviceNotFoundError: I2C device not found on address: 0x3C
```

Could also mean that your I2C isn't enabled properly as well.  But let's check the connections of the display first.

Chadac77, 2021-11-04:
yes i will hook display i had it disconnected for testing trying to see if it was the problem along with most everything. I will hook up everything and run it and then copy the file.

Chadac77, 2021-11-04:
ok I saved the new files in the drive folder

nebhead, 2021-11-04:
OK, I think the screen might be OK now, but it's hard to tell.  Can you delete the `control.err.log` file in the logs directory to start it fresh?   

It appears that there may be an issue with the ADS1115 (ADC) readings, as I see the output of the `control.out-new.log`: 
```
2021-11-01 13:16:34 Error Reading Temperature.
```

Maybe double check you have this hooked up correctly as well. 

Also, for any files that you don't have permission to change, you can always use `sudo`.  

For example:

`sudo cat control.err.log`
`sudo rm control.err.log`

Chadac77, 2021-11-05:
it seems to be  working now 
i will test it 
thank you so much

nebhead, 2021-11-05:
You're welcome!  Enjoy!

closed 2021-11-05 by nebhead