nebhead/PiFire #74: No grill temp/ADS

ryanmurray, 2022-01-22:
So i assembled my prototype. Screen turns on. and fan, auger, hot rod turn on. Unfortunately im unable to get grill temp readings. Im using the pcb and have attached the ads chip with VDD meeting 3.3v spot. I2C is enabled, baud rate added to line, and core frequency added to end. Logs just tell me that temperature was unable to be read. Where should i start :/

ryanmurray, 2022-01-23:
Nevermind checked all solder joints and everything and got a reading. Test fire revealed hot rod appears dead. Considering my traegar died and i have voltage coming to the plug id say this is why it broke before :P

nebhead, 2022-01-24:
Oh man... I think I've gone through 3 hot rods over the years with my Traeger.  They just aren't very durable.  :(  Anyway, good luck.

closed 2022-01-24 by nebhead