nebhead/PiFire #100: ili9341_encoder.py Encoder and display do not work.

brandescm, 2022-05-14:
When selecting this option during the wizard it leaves the screen and encoder non-functional. 

**To Reproduce**
1. Run wizard from admin menu
2. Select ili9341_encoder option

**Expected behavior**
Display will show properly and the encoder will work. 

**Screenshots**

**PiFire Version:**
 - Version latest

**Additional context**
ili9341_encoder.py changes the mapping of the display pins and changes the backlight pin. Here is the fix if you want to keep the wiring the same between modules and have encoder and button functionality. I can certainly create a branch and create a pull request if you like. Otherwise here is the fix I used in ili9341_encoder.py.
Line43
		`# spi set to make the display work with other spi devices like a max31865
		self.serial = spi(port=0, device=0, gpio_DC=24, gpio_RST=25, bus_speed_hz=32000000,
						  reset_hold_time=0.2, reset_release_time=0.2)`
Line 81
		`clk = 22
		dt = 23
		sw = 12`

		`self.my_encoder = pyky040.Encoder(clk, dt, sw)
		self.my_encoder.setup(scale_min=0, scale_max=100, step=1, inc_callback=self.inc_callback,
							  dec_callback=self.dec_callback, sw_callback=self.click_callback)`

nebhead, 2022-05-15:
Are you using the Main branch or the Development branch?  In the development branch, I'd just recently committed a new file called display_ili9341e.py which maintains the same SPI pinout and button board wiring.  The file that you mention above (ili9341_encoder.py) was user submitted and was meant for their specific use case while using an additional MAX31865 on the SPI bus at the same time. 

That said, the entire display architecture is going to be getting a big overhaul soon, and I will be renaming the above file to follow the naming convention, and will provide some more detail about the pinout in the description of the wizard installer.  Hopefully will give some clarity to users when selecting the right option in the wizard.  

The encoder is a great option and I'm going to be installing it in my personal rig very soon.

brandescm, 2022-05-15:
I would have been on main for this. I'll pull develop and take a look. 

The whole cook cycle needs a finite state machine and each cook step broken out. 

Good luck with the updates

nebhead, 2022-05-15:
> I would have been on main for this. I'll pull develop and take a look.

Great!  Let me know how it goes.  Maybe just confirm that the code has the right pinout for your setup.  

> The whole cook cycle needs a finite state machine and each cook step broken out.

This is an interesting idea.  Admittedly, the work cycle in PiFire has been a bit of a messy bit of code.  And code is duplicated for the manual and monitor modes (which is not ideal).  I would be interested to understand a bit more about how this could help structure things.  

> Good luck with the updates

Thanks!

brandescm, 2022-05-15:
> This is an interesting idea. Admittedly, the work cycle in PiFire has been a messy bit of code. And code is duplicated for the manual and monitor modes (which is not ideal). I would be interested to understand a bit more about how this could help structure things.

I have a separate bitbucket repo with these changes. Basically cloned your project and started making this update. Right now it is private but can give you access  if wanted to look at it.

nebhead, 2022-05-15:
Yeah, I would love to take a look.  I'm open to anything that can make this project even better, and learning something in the process.  My username over at bitbucket is nebhead.

brandescm, 2022-05-15:
> Yeah, I would love to take a look. I'm open to anything that can make this project even better, and learning something in the process. My username over at bitbucket is nebhead.

Name isn't coming up, not sure if bitbucket is the issue or what. I set the repo to public (piFire) brandescm is my username over there.

nebhead, 2022-07-18:
FYI, I'm closing this one as the latest updates to the main branch have working ILI9341 Encoder support.

closed 2022-07-18 by nebhead