nebhead/PiFire #105: ILI9341 Screen rotation 90 degrees does not work

NessK, 2022-07-17:
**Describe the bug**
ILI9341 240x320 display doesnt work when display rotation is set to 90 degrees in the wizard. The backlight is on, but nothing is displayed.
I have tried 0 and 180 during troubleshooting, then it works as expected, but nothing is displayed when 90 degrees is selected.
I have tried the setting "ILI9341 TFT with 3-button Input" and "ILI9341 TFT" in the wizard, results are the same. it works in landscape mode, but not portrait mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Wizard form Admin Menu
2. Click on Display
3. Select ILI9341 TFT
4. Select 90 degrees

**PiFire Version:**
 - Version 1.3.4
 - Branch Main

Thanks for all the great work in this awesome project! Im waiting for my PCB to arrive, cant wait to test it all! 
I will post in the build thread when I get the PCB.

nebhead, 2022-07-17:
Hey @NessK, glad you're enjoying the project!  Looking forward to seeing your build. 

You have found a bug. :)  With that said, it may takes some time to fix this because the display module currently isn't designed to support 90 degrees due to the pixel width being smaller in that dimension.  Many of the screen drawing elements may fail in that scenario.  Perhaps in the future, we can try to update the module to support this rotation angle.  For now, I'm going to remove the rotation angle of 90 degrees from the wizard so that folks don't run into this issue again.  Thanks for bringing it to my attention!

nebhead, 2022-09-25:
FYI, I'm going to work on 90 degree rotation soon.  In the development branch, I've refactored the display code for 240x320 displays to make it easier to deploy this more globally.  Likely going to tackle this next week.

nebhead, 2022-09-29:
90 & 270 Degree Rotations have been added back into the 240x320 display modules in the development branch: c3a1f2a1c4fad622b36232d4800e8e3078cd40e1  

When it's thoroughly tested, we'll merge it into the Main for the next release.   With that, I'm going to close this particular feature request.  Thanks!

closed 2022-09-29 by nebhead