nebhead/PiFire #121: Apprise Notification Integration

calonmerc, 2022-09-23:
**Is your feature request related to a problem? Please describe.**
I exclusively use Discord and Slack on my phone, not Pushbullet or Pushover

**Describe the solution you'd like**
Switch to using [Apprise](https://github.com/caronc/apprise)

**Additional context**
I'm going to attempt to get this to work, then submit a pull request, but won't be able to start until I find an extra RPi (tomorrow I'll attempt on an OrangePi, just to get developing)

Thanks a ton for all your work on this, I'm beyond happy I found the project!

nebhead, 2022-09-23:
Wow, thanks for letting me know about Apprise, looks like a great solution to offer many different notification options.  I like it.  I will check it out.  

If you do work on this on your own, please do make sure you are on the development branch and using the latest.  Thanks!!  Otherwise, I will take a look and pull it in at some point if it looks good.

calonmerc, 2022-09-23:
Apprise is pretty great from a developer's perspective, as you only have to support their integration and a user can decide to use whatever notification system(s) they want! I've never personally integrated Apprise, but I've worked on a few projects that have used it.

On that note, I'd love to be able to support multiple notification destinations, which Apprise supports natively, just concat the destinations in the command (in the CLI, at least). In the Python implementation, I believe it's `AppriseObject.add('<provider>://<auth>')`

nebhead, 2022-10-03:
Just wanted to say thanks again for the addition of this feature to the project.  I've tested it a bit and it works well for pushover, though it would be cool to setup local-URL support.  There are a few minor corner cases if you accidentally delete all of the list items, but I'm not going to mess with it right now.  In the future, I feel like it would be good to replace all of the other notification methods with Apprise so that we can simplify things a bit.  Again, that will be for another time. 

Also, I've learned a few new things from your JS and Flask changes, particularly how to input list of items which is great!  So this is a double thank you!

closed 2022-10-03 by nebhead

weberbox, 2022-10-09:
I just wanted to mention here that unfortunately the OneSignal implementation in Apprise will not work due to it requiring the API key to send notifications. I cannot commit the api key to Github as that is pretty much the password for the OneSignal account and with it anyone could do some serious damage. 

Currently using the post method it works because if you are sending directly to a player id then the api key is not required. 

Maybe some day I can take a look at the Apprise code and submit a pr to them so when only a player id is used the api key is not required.