nebhead/PiFire #123: Installing on Orange Pi Zero 2

calonmerc, 2022-09-26:
**Describe the bug**
502 bad gateway when trying to view main web page. Supervisor web UI, `Tail -f Stderr` for the webapp repeats this error every second:
```Error: class uri 'eventlet' invalid or not found: 

[Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gunicorn/util.py", line 99, in load_class
    mod = importlib.import_module('.'.join(components))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/gunicorn/workers/geventlet.py", line 10, in <module>
    import eventlet
  File "/usr/local/lib/python3.10/dist-packages/eventlet/__init__.py", line 17, in <module>
    from eventlet import convenience
  File "/usr/local/lib/python3.10/dist-packages/eventlet/convenience.py", line 7, in <module>
    from eventlet.green import socket
  File "/usr/local/lib/python3.10/dist-packages/eventlet/green/socket.py", line 4, in <module>
    __import__('eventlet.green._socket_nodns')
  File "/usr/local/lib/python3.10/dist-packages/eventlet/green/_socket_nodns.py", line 11, in <module>
    from eventlet import greenio
  File "/usr/local/lib/python3.10/dist-packages/eventlet/greenio/__init__.py", line 3, in <module>
    from eventlet.greenio.base import *  # noqa
  File "/usr/local/lib/python3.10/dist-packages/eventlet/greenio/base.py", line 32, in <module>
    socket_timeout = eventlet.timeout.wrap_is_timeout(socket.timeout)
  File "/usr/local/lib/python3.10/dist-packages/eventlet/timeout.py", line 166, in wrap_is_timeout
    base.is_timeout = property(lambda _: True)
TypeError: cannot set 'is_timeout' attribute of immutable type 'TimeoutError'
]
```

**PiFire Version:**
 - 1.3.1

**Additional context**
Confirmed that all apt packages installed properly and all pip3 packages installed properly. Verified eventlet version 0.30.2

calonmerc, 2022-09-27:
Fixed that issue by installing eventlet from master, apparently there is a bug not yet released relating to Python 3.10.

Now I get a different error:
```Error: class uri 'eventlet' invalid or not found: 

[Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gunicorn/util.py", line 99, in load_class
    mod = importlib.import_module('.'.join(components))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/gunicorn/workers/geventlet.py", line 10, in <module>
    import eventlet
  File "/usr/local/lib/python3.10/dist-packages/eventlet/__init__.py", line 17, in <module>
    from eventlet import convenience
  File "/usr/local/lib/python3.10/dist-packages/eventlet/convenience.py", line 7, in <module>
    from eventlet.green import socket
  File "/usr/local/lib/python3.10/dist-packages/eventlet/green/socket.py", line 21, in <module>
    from eventlet.support import greendns
  File "/usr/local/lib/python3.10/dist-packages/eventlet/support/greendns.py", line 66, in <module>
    setattr(dns, pkg, import_patched('dns.' + pkg))
  File "/usr/local/lib/python3.10/dist-packages/eventlet/support/greendns.py", line 61, in import_patched
    return patcher.import_patched(module_name, **modules)
  File "/usr/local/lib/python3.10/dist-packages/eventlet/patcher.py", line 132, in import_patched
    return inject(
  File "/usr/local/lib/python3.10/dist-packages/eventlet/patcher.py", line 109, in inject
    module = __import__(module_name, {}, {}, module_name.split('.')[:-1])
  File "/usr/local/lib/python3.10/dist-packages/dns/namedict.py", line 35, in <module>
    class NameDict(collections.MutableMapping):
AttributeError: module 'collections' has no attribute 'MutableMapping'
]
```

From research, appears to be a change to the collections library, as mentioned in [this comment in the Medusa github](https://github.com/pymedusa/Medusa/issues/10253#issuecomment-1015871418)

nebhead, 2022-09-27:
I was having similar issues in my development environment on PopOS! however after switching back to Ubuntu, and just installing the vanilla versions of everything, it started to work just fine.  I never did figure out what the issue was, but I can show you what my Ubuntu environment looks like: 

```
Python 3.10.6 (main, Aug 10 2022, 11:40:04) [GCC 11.3.0] on linux
```

```
ben@framework:~$ sudo pip list

Package                Version
---------------------- ---------------
apturl                 0.5.2
async-timeout          4.0.2
bcrypt                 3.2.0
bidict                 0.22.0
blinker                1.4
Brlapi                 0.8.3
caffeine               2.9.8
certifi                2020.6.20
chardet                4.0.0
ciso8601               2.2.0
click                  8.0.3
colorama               0.4.4
colorzero              2.0
command-not-found      0.3
cryptography           3.4.8
cupshelpers            1.0
dbus-python            1.2.18
defer                  1.0.6
Deprecated             1.2.13
distro                 1.7.0
distro-info            1.1build1
dnspython              2.2.1
duplicity              0.8.21
eventlet               0.33.1
ewmh                   0.1.6
fasteners              0.14.1
Flask                  2.2.2
Flask-Mobility         1.1.0
Flask-QRcode           3.1.0
Flask-SocketIO         5.3.1
future                 0.18.2
gpiozero               1.6.2
greenlet               1.1.3
httplib2               0.20.2
idna                   3.3
importlib-metadata     4.6.4
influxdb-client        1.32.0
itsdangerous           2.1.2
jeepney                0.7.1
Jinja2                 3.1.2
keyring                23.5.0
language-selector      0.1
launchpadlib           1.10.16
lazr.restfulclient     0.14.4
lazr.uri               1.0.6
lockfile               0.12.2
louis                  3.20.0
macaroonbakery         1.3.1
Mako                   1.1.3
MarkupSafe             2.1.1
monotonic              1.6
more-itertools         8.10.0
netifaces              0.11.0
oauthlib               3.2.0
olefile                0.46
packaging              21.3
paramiko               2.9.3
pexpect                4.8.0
Pillow                 9.0.1
pip                    22.0.2
protobuf               3.12.4
ptyprocess             0.7.0
pycairo                1.20.1
pycups                 2.0.1
pygame                 2.1.2
PyGObject              3.42.1
PyJWT                  2.3.0
pymacaroons            0.13.0
PyNaCl                 1.5.0
pyparsing              2.4.7
pyRFC3339              1.1
python-apt             2.3.0+ubuntu2.1
python-dateutil        2.8.1
python-debian          0.1.43ubuntu1
python-engineio        4.3.4
python-socketio        5.7.1
python-xlib            0.29
pytz                   2022.1
pyxdg                  0.27
PyYAML                 5.4.1
qrcode                 7.3.1
reactivex              4.0.4
redis                  4.3.4
reportlab              3.6.8
requests               2.25.1
SecretStorage          3.3.1
setuptools             59.6.0
six                    1.16.0
systemd-python         234
typing_extensions      4.3.0
ubuntu-advantage-tools 27.10
ubuntu-drivers-common  0.0.0
ufw                    0.36.1
unattended-upgrades    0.1
urllib3                1.26.5
usb-creator            0.3.7
uuid                   1.30
wadllib                1.3.6
Werkzeug               2.2.2
wheel                  0.37.1
wrapt                  1.14.1
xdg                    5
xkit                   0.0.0
zipp                   1.0.0
```

Maybe this will help.

calonmerc, 2022-09-27:
Odd... I'll compare, running Ubuntu 22.04, Python 3.10.6

```sudo pip list
Package                Version
---------------------- ---------------
async-timeout          4.0.2
bidict                 0.22.0
blinker                1.4
certifi                2022.9.24
chardet                4.0.0
charset-normalizer     2.1.1
ciso8601               2.2.0
click                  8.1.3
colorzero              2.0
cryptography           3.4.8
dbus-python            1.2.18
Deprecated             1.2.13
distro                 1.7.0
distro-info            1.1build1
dnspython              1.16.0
eventlet               0.33.1
Flask                  2.2.2
Flask-Mobility         1.1.0
Flask-QRcode           3.1.0
Flask-SocketIO         5.3.1
gpiozero               1.6.2
greenlet               1.1.3
gunicorn               20.1.0
httplib2               0.20.2
idna                   3.4
importlib-metadata     4.6.4
influxdb-client        1.32.0
iotop                  0.6
itsdangerous           2.1.2
jeepney                0.7.1
Jinja2                 3.1.2
keyring                23.5.0
launchpadlib           1.10.16
lazr.restfulclient     0.14.4
lazr.uri               1.0.6
MarkupSafe             2.1.1
more-itertools         8.10.0
netifaces              0.11.0
oauthlib               3.2.0
packaging              21.3
Pillow                 9.0.1
pip                    22.0.2
pushbullet.py          0.12.0
PyGObject              3.42.1
PyJWT                  2.3.0
pyparsing              2.4.7
python-apt             2.3.0+ubuntu2.1
python-dateutil        2.8.2
python-debian          0.1.43ubuntu1
python-engineio        4.3.4
python-magic           0.4.27
python-socketio        5.7.1
PyYAML                 5.4.1
qrcode                 7.3.1
reactivex              4.0.4
redis                  4.3.4
requests               2.28.1
RPi.GPIO               0.7.1a4
SecretStorage          3.3.1
setuptools             65.4.0
six                    1.16.0
supervisor             4.2.1
typing_extensions      4.3.0
ubuntu-advantage-tools 27.10
unattended-upgrades    0.1
urllib3                1.26.12
uuid                   1.30
wadllib                1.3.6
websocket-client       1.4.1
Werkzeug               2.2.2
wheel                  0.37.1
wrapt                  1.14.1
zipp                   1.0.0
```

I'm only looking at the Orange Pi as a development environment, as it's limited in GPIO (and they're all different, from my first glance).

calonmerc, 2022-09-27:
!!!! I might have found it. More research appears to point to the `dnspython` package. In your list it's 2.2.1, whereas mine is 1.16.0.

Attempting to update now. `sudo pip3 install git+https://github.com/rthalley/dnspython.git`

And a different error. I'm going to keep updating with my debugging, in case anyone else falls into a similar place. 

```python3.10/dist-packages/eventlet/green/socket.py", line 21, in <module>
    from eventlet.support import greendns
  File "/usr/local/lib/python3.10/dist-packages/eventlet/support/greendns.py", line 66, in <module>
    setattr(dns, pkg, import_patched('dns.' + pkg))
  File "/usr/local/lib/python3.10/dist-packages/eventlet/support/greendns.py", line 61, in import_patched
    return patcher.import_patched(module_name, **modules)
  File "/usr/local/lib/python3.10/dist-packages/eventlet/patcher.py", line 132, in import_patched
    return inject(
  File "/usr/local/lib/python3.10/dist-packages/eventlet/patcher.py", line 109, in inject
    module = __import__(module_name, {}, {}, module_name.split('.')[:-1])
  File "/usr/local/lib/python3.10/dist-packages/dns/zone.py", line 86, in <module>
    class Zone(dns.transaction.TransactionManager):
  File "/usr/local/lib/python3.10/dist-packages/dns/zone.py", line 757, in Zone
    ) -> dns.rdtypes.ANY.SOA.SOA:
AttributeError: module 'dns.rdtypes' has no attribute 'ANY'
]
```

nebhead, 2022-09-28:
Maybe instead try: 

```bash
pip3 install --upgrade dnspython
```

This has worked for me in the past.

calonmerc, 2022-09-30:
I've debugged for a few days and am coming to a similar conclusion as you... Probably going to steal one of my Raspberry Pis from one of my 3D Printers 😂

closed 2022-09-30 by calonmerc