nebhead/PiFire #132: Control Crashing on write_cookfile

weberbox, 2022-10-09:
**Describe the bug**

Control crashes in manual mode once stopped with an error stemming from write_cookfile()

```
Traceback (most recent call last):
  File "/usr/local/bin/pifire/control.py", line 1033, in <module>
    write_cookfile()
  File "/usr/local/bin/pifire/common.py", line 1350, in write_cookfile
    starttime = json.loads(historydata[0])
IndexError: list index out of range
```

**To Reproduce**
Steps to reproduce the behavior:
1. Start Manual Mode
2. Stop Manual Mode

**Expected behavior**

No Crash :)

**PiFire Version:**
 - Development 1.3.5

**Additional context**

Not sure if other modes are also crashing I have only noticed on Manual Mode. 

My temporary fix was to just check the history data to avoid the IndexError

```
historydata = cmdsts.lrange('control:history', 1, -1)

if not historydata:
     return

```

Sorry I have not had much time to read through all the new cookfile stuff so I do not fully understand how it works yet

nebhead, 2022-10-09:
No worries - I will look into this.  So far, I haven't seen this error before, but I haven't used Manual mode since implementing this so there may be an issue there.  Anyway, I will take a look and see what's going on.

nebhead, 2022-10-09:
Pushed a fix - https://github.com/nebhead/PiFire/commit/d230781de9ddaa28e0ea3caad9bad93da7c54e43

Give that a try for me.

weberbox, 2022-10-10:
Seems to have fixed it. Easy enough lol, I should have sent a PR instead of taking up your time. Thanks for taking a look.

closed 2022-10-10 by weberbox

nebhead, 2022-10-10:
> Seems to have fixed it. Easy enough lol, I should have sent a PR instead of taking up your time. Thanks for taking a look.

Ha!  I mean, it was an easy fix but somewhat complex bug. :P Anyway, never feel bad about submitting an issue - it's my pleasure to investigate and fix.  Take care and have a great week!