nebhead/PiFire #137: Failed Startup Error

weberbox, 2022-10-24:
**Describe the bug**
After grill failed to ignite/startup then failed to reignite an error was shown on screen. Problem is after a couple seconds the error cleared and the screen went back to reignite but nothing was actually running. 

Unfortunately it seems my igniter took a crap so the grill never actually fired up and this bug presented itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Grill
2. Grill does not meet startup temp
3. Reignite is tried
4. Reignite fails
5. Error is displayed
6. Screen goes back to Reignite

**Expected behavior**
Should stay on error screen I believe 

**PiFire Version:**
 - v1.3.6 Development 

**Additional context**
Creating this issue so I do not forget. I am not sure I can look into this until next week and by then I will probably forget this ever happened lol

nebhead, 2022-10-24:
Oh no!  I've had several igniters fail on me - the seem to be so dang cheap and easily broken.  I wonder if anyone makes any more robust igniters that would last longer.  Anyway, I'll keep this one in the back of my mind, and we'll see who can get to it first. :P

nebhead, 2022-11-16:
I think this should be fixed with one of the last updates.  In an error state, we were keeping the screen on to show the error message, but we never cleared it afterwards which means the screen stays in an active state.  I simply introduced a short delay before clearing the screen when an Error state is selected.  Anyway, give it a try and let me know if you are able to reproduce anymore.

weberbox, 2023-01-23:
I think this is fixed. Will close this issue and open a new one if it crops up again in the future.

closed 2023-01-23 by weberbox