nebhead/PiFire #143: Auger feed rate multipler

ChasonDeshotel, 2022-11-15:
"the PID calc does not take into consideration the auger feed rate. I assume the PID calcs were created for a given feed rate which likely does not apply for all appliances, like mine, for instance. If there were a multiplier or divisor that took into account the feed rate of the auger, I think PID tuning would be easier"
-FastEddie49

nebhead, 2022-11-16:
Yep, I think this one needs more investigation, given we are all sort of trying to figure out how all of this can be modified to get the proper results out.

FastEddie49, 2022-11-20:
I have since come to the conclusion that, in the handful of times I have used the autotune feature of temperature controllers at work, that PIDs don't need to know things like pump output in gallons/min or weights of things being heated. It's not needed to know the specifics of a system ("plant") for PIDs to work with it, the same way that PIDs don't need to be told if the application is a heating or pumping operation. So scratch that idea.

As to the second part of this, I took 2 measurements of the auger output at 20 seconds each and averaged the 2 to come up with 0.8 grams/second output. I set the feed rate in the program accordingly. I then ran a cook of just under 2 hours and did the math as described earlier. My actual pellet feed averaged over the cook time is 0.825, pretty darn close to what I achieved by using the more simple method. I did up my feed rate to 0.85 in the program because I'd rather have an overestimate than an under, but that's personal preference. I can validate that the simple method is accurate enough for most people.

nebhead, 2023-06-05:
I'm going to close this one out, as I'm probably not going to spend the time to update the PID.  However, with the release of v1.5.0 I welcome others to dive into the controller and hack it to their hearts content. :)

closed 2023-06-05 by nebhead