nebhead/PiFire #149: More meat temp probes

kpalu, 2023-02-16:
**Is your feature request related to a problem? Please describe.**
A lot of the new high end pit controls have 4 temp probes and fireboard has 6. In a situation where you have a large upright smoker it would be nice to have the ability to have 6-8 meat probes going at once. 

**Describe the solution you'd like**
My idea would be a usb daughter card that could have the extra temp channels on it. Something that if you plugged it in it might disable the two probes inputs on the main daughter card. 

**Describe alternatives you've considered**
I've seen a 10 channel usb input card around on a few sights (I haven't done enough research to know if it could be used with rtd's) for around $15. Something in that form factor is what I think we would need. 

**Additional context**
The card i was looking at is on Amazon 
Asin# ‎B07ZCWRK4H
Thank you

nebhead, 2023-02-16:
You are in luck!  I've pushed a version of PiFire to the development branch that incorporates basically an unlimited amount of probes (well, limited by the Raspberry Pi capacity).  However, these are basically allowing you to stack the existing probe input devices (i.e. ADC1x15 boards, MAX38165 boards).  So you'd still need to have multiple ADC boards and either design a PCB to utilize these additional boards or just wire them up on a breadboard.  If you don't feel comfortable with designing a PCB, I'm sure there would be someone on the PiFire discord community who would be interesting in giving it a shot.  

I'm going to take a look at the USB based ADC that you listed.  That's very interesting indeed.  Thanks for sharing it!  

Anyway, if you are brave enough, you can check the version on the development branch.  (note, your settings will be modified for the new version, so if you ever want to go back, you'll want to make sure you've backed up)  I'm waiting for more testing on it before I take it to the production/main branch, as well as the developer of the Android application to update the app to support the new version.

faisalhusain007, 2023-02-19:
Hi @nebhead sorry been tied up for a while, I use a cheap ibbq type bluetooth thermometer with 6 probe temps comes as complete set for like $20 if you look around. Uses BLE the libraries are well documented I use it for other cooks on my weber etc and use the rpi running pifre as a gateway to upload to my mqqt server where I can check on cooks. I also use it for cooks on the pifire. This gives me a lot of flexibility, I was thinking if we can integrate this into Pifire as the cheapest possibly  easy solution. Anyway here are some examples of implementation and the ibbq lib  https://github.com/adafruit/Adafruit_CircuitPython_BLE_iBBQ
Here is a link running stand alone barebones without Adafruit lib 
https://github.com/8none1/pybq

nebhead, 2023-06-05:
> Hi @nebhead sorry been tied up for a while, I use a cheap ibbq type bluetooth thermometer with 6 probe temps comes as complete set for like $20 if you look around. Uses BLE the libraries are well documented I use it for other cooks on my weber etc and use the rpi running pifre as a gateway to upload to my mqtt server where I can check on cooks. I also use it for cooks on the pifire. This gives me a lot of flexibility, I was thinking if we can integrate this into Pifire as the cheapest possibly easy solution. Anyway here are some examples of implementation and the ibbq lib https://github.com/adafruit/Adafruit_CircuitPython_BLE_iBBQ Here is a link running stand alone barebones without Adafruit lib https://github.com/8none1/pybq

I don't know how I missed this for so long, I definitely want to look into this!   

For now, with version 1.5.0 being released today, it is now possible to add more probes to the configuration (if your hardware supports it of course).  I'll go ahead and close this one, but I think a module to support bluetooth is a really cool idea.

closed 2023-06-05 by nebhead