nebhead/PiFire #154: PiFire setup wizard does not update options properly

ghandye, 2023-08-06:
**Describe the bug**
When using the PiFire setup wizard, and there are multiple options to configure on each screen (e.g. platform, display, etc.), the options revealed no longer change when selecting different platforms/displays. Similarly, the picture of the selected display does not update. It almost seems like whatever background data load / page change is supposed to be happening is broken.

**To Reproduce**
Steps to reproduce the behavior:
0. Install fresh OS / PiFire (I used PiFireOS nightly 0.2.0 2023-08-05 which is tied to development branch 1.6.0)
1. Admin > Config Wizard > Launch
2. Next
3. On Platform screen, click between Standard / PWM / Prototype
4. BUG - note how options section does not change dynamically when you change platforms
5. Next again (Probe Input)
6. Next again (Display)
7. On Display screen, select ST7789 + Rotary Encoder from the drop-down
8. BUG - note how options section does not change dynamically when you change displays

Browser: Firefox 116.0.1 / Windows 10

**Expected behavior**
Options sections should dynamically update when you change the main item in the platforms / displays section. 

Note that this worked correctly on PiFireOS 0.1.0 / PiFire 1.5.2.

**Screenshots**
[image]
[image]

**PiFire Version:**
 - Version 1.6.0

ghandye, 2023-08-06:
Some additional data points that also fail:
- I wiped the Pi Zero W and followed the PiFire guide, installing vanilla Raspberry Pi OS Lite and running the install script, bug still present
- Tried the above with both Firefox and Microsoft Edge, same behavior

nebhead, 2023-08-07:
Looking into this... wanted to ask if you had wiped your history/cache before trying the wizard again?  I have seen some similar behavior in the past with Firefox.  :/

nebhead, 2023-08-07:
OK, I reproduced the issue.  Looks like a bug with my code, not sure how the heck I missed this.  I'll get it fixed and pushed out soon.  Thanks for the catch!

ghandye, 2023-08-07:
That’s great news! I did encounter the bug on a different browser as well, so it didn’t seem to be a session or cache issue. What’s weird is, it worked at least once for me historically, otherwise I wouldn’t have recognized it was broken.

nebhead, 2023-08-08:
OK, I have updated the code and I think have resolved the issue. I quickly tested on a bug-fix branch and it all looks good on my end.  So I went ahead and pushed to main (https://github.com/nebhead/PiFire/commit/f44a651194a2c579acaa3096d2b17bbf8bf32c93) which you can test out now.  Hopefully it's all cleared up.  Thanks again for submitting the issue!

ghandye, 2023-08-08:
Looks good!

Tested again on 1.5.2 (pre-fix) with Firefox in a private window - reproduced problem immediately.
Updated to 1.5.3, let the app restart, tested again - problem did not occur.

Thanks for the fix!

closed 2023-08-08 by ghandye