nebhead/PiFire #165: Food Temp is Irratic if there isn't a Grill Temp Plugged In

EricPalmquist, 2023-10-25:
While bench testing, without the Grill probe (which is A1 for me), the Food temp (which is in A2) is verry erratic, sometimes showing the correct temp (71), sometimes showing 0, and sometimes showing just 7.  At first I thought the 7 was random, but then noticed that when the temp dropped down to 69 is read 6, so it appears that it is an issue with displaying all the digits.

To reproduce, configure A0=Food probe, A1=Grill probe, and A2=Food probe 2.  Connect the food probes but leave the grill probe disconnected, and monitor the dashboard.

Disconnecting one probe should not impact the others.

[image]

Version 1.5.4

nebhead, 2023-10-26:
Thanks for the bug submission.  I'll try to take a look at this soon on my bench test rig and see if I can reproduce.  I have a theory about the 7F and 6F readings.  The temperature is stored in a queue (to prevent erroneous sensor/ADC inputs) which happens to be 10 elements in length if the queue has [73, 0, 0, 0, 0, 0, ... 0] the average of that queue will be 7.3 or 7 for the sake of this output.  I think that is what we may be seeing here, but I could always be way off base.  

Another thing to consider is your probe types.  Let me know what kind of probes you are using for the grill probe and each of the food probes.  

Also check your solder points, that the ADC1115 is properly seated and your wiring is solid.  

Thanks!

nebhead, 2023-10-27:
Unable to reproduce on my end.  I've disconnected the grill probe and am running two food probes (they are actually pot's on a test board).  No deviation in temperatures for the food probes in this scenario. 

[image]

Probe Profiles selected:

[image]

Release v1.5.4

[image]

Next steps: 
1. As mentioned int he previous response, let me know what probes and probe profiles you are using for your food probes.  
2. Check all of your wiring, your solder points, screw terminals, etc.  Check your ADS1115 is properly seated (if socketed), etc.  

Thanks!

Eric-Palmquist-FHR, 2023-11-05:
At the time I was doing that testing I had the probes set to use the PT-1000-Grill-Probe-PiFire profiles for all 3.  Since then, I've moved to the v1.6.0-dev40 branch and now have everything bolted up inside the hopper so it's not very easy for me to try to reproduce.  I feel good about the mechanical mounting.  I actually soldered in the ADS1115.  I'm now troubleshooting another odd issue with may be related to this, but I likely should open as a new issue.  Let me know what you think.  

To get my three temps reading reliably (food1 food2 and grill1) I found that in the config wizard I needed to also define the unused Grill2 probe.  I assigned it to the Axillary type so that it isn't displayed anywhere.  Should I need to define unused one?  What I see now is that Grill 1 reads great most of the time, but sometimes momentarily drops to 0.  To troubleshoot this I've been running the probe tuner (which is really cool by the way) to see the raw resistance reading.  What I see here is that Grill1 gets reasonable readings like 1106, and then occasionally shoots up to around 200,000.  But here is what gets really strange.  When I use the profiler to read the resistance on the un-used Grill2 input I see somewhat the opposite - it reads around 200000 most of the time, but occasionally jumps to 1106.  So it seems like somewhere the inputs from the ADS are getting mixed up.

Here is my config:
[PiFire_11-05-23_160157.json](https://github.com/nebhead/PiFire/files/13260898/PiFire_11-05-23_160157.json)

nebhead, 2023-11-06:
> I'm now troubleshooting another odd issue with may be related to this, but I likely should open as a new issue. Let me know what you think. To get my three temps reading reliably (food1 food2 and [...]

Short answer is no, you shouldn't have to define the unused port.  In fact, there are others here that don't have food probes defined at all.  So you should be able to reliably just use one probe or a subset of the ports on the ADC without issue.  I personally run this all the time with both my bench test setup and with my actual grill.  No issues thus far.  

> What I see now is that Grill 1 reads great most of the time, but sometimes momentarily drops to 0. To troubleshoot this I've been running the probe tuner (which is really cool by the way) to see the [...]

Very odd behavior, but I think we might have a fix for this.  I noticed in your settings that you are using the vanilla ADS1115 module, which several folks have reported issues with.  Let's try to use the Adafruit module in the configuration wizard instead of the old ADS1115 module.  This should be the default module, but perhaps if you are coming from an older setup, it might not be.  

Note the device 'type': 
[image]

Add the new device:
[image]

Make sure you have your device configured properly here. 

Then make sure you assign your ports to the right probes, also ensure you delete the old ADS1115 device.  

Give that a go.

EricPalmquist, 2023-11-06:
I get this when I try the ads1115_adafruit module (with V 1.6.0-dev40)
2497	2023-11-06 12:55:08 [ERROR] An error occurred loading the [ads1115_adafruit] probe module. The prototype module has been loaded instead. This sometimes means that the hardware is not connected properly, or the module is not configured. Please run the configuration wizard again from the admin panel to fix this issue.
2496	2023-11-06 12:54:53 [ERROR] The control process experienced a timeout event (no heartbeat detected in 30 seconds) and is being reset.
2495	2023-11-06 12:54:23 [INFO] Control Script Starting Up.
2494	AttributeError: module 'adafruit_platformdetect' has no attribute 'Detector'
2493	detector = adafruit_platformdetect.Detector()
2492	File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/agnostic/__init__.py", line 20, in <module>
2491	from adafruit_blinka.agnostic import board_id, detector
2490	File "/usr/local/lib/python3.9/dist-packages/board.py", line 22, in <module>
2489	import board
2488	File "/usr/local/bin/pifire/probes/ads1115_adafruit.py", line 36, in <module>

nebhead, 2023-11-06:
Very strange.  Can you try rebooting again to see if this still occurs?  We may need to re-installed the Adafruit module as it seems like it's crashing for some reason.

EricPalmquist, 2023-11-06:
Reboot didn't help.  I've tried removing and adding back the ADS1115_adafruit module in the wizard multiple times.  Is there a way to manually re-install the Adafruit module?  BTW - here is the output when I run the wizard.  Are these the correct versions?

Requirement already satisfied: adafruit-circuitpython-ads1x15 in /usr/local/lib/python3.9/dist-packages (2.2.23)
Requirement already satisfied: typing-extensions~=4.0 in /usr/local/lib/python3.9/dist-packages (from adafruit-circuitpython-ads1x15) (4.8.0)
Requirement already satisfied: adafruit-circuitpython-typing in /usr/local/lib/python3.9/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.9.5)
Requirement already satisfied: Adafruit-PlatformDetect>=3.53.0 in /usr/local/lib/python3.9/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-ads1x15) (3.53.0)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in /usr/local/lib/python3.9/dist-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (3.5)
Requirement already satisfied: adafruit-circuitpython-requests in /usr/local/lib/python3.9/dist-packages (from adafruit-circuitpython-typing->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (2.0.2)

nebhead, 2023-11-06:
> Reboot didn't help. I've tried removing and adding back the ADS1115_adafruit module in the wizard multiple times. Is there a way to manually re-install the Adafruit module? 

You could try to SSH into your pi and then attempt to do the following:

```bash
sudo python -m pip install --upgrade adafruit-circuitpython-ads1x15
```
To be honest, I'm not sure if this is going to work given the output of the above.  There may be something else of note going on.  Have you changed the I2C address of your device?

nebhead, 2023-11-06:
Also, what Raspberry Pi board and OS are you currently running on?  From the previous error messages, it looks like the Adafruit libraries can't detect the platform correctly, which is a brand new error which I haven't seen before.

EricPalmquist, 2023-11-06:
Tried pip uninstall, reinstall, etc. but always get that same error.  I had ChatGPT create a python script for me to test with, which also get the same error.  I am running on
Model : Raspberry Pi Zero W Rev 1.1

nebhead, 2023-11-06:
> Tried pip uninstall, reinstall, etc. but always get that same error. I had ChatGPT create a python script for me to test with, which also get the same error. I am running on is below. Also no, I didn't change the address.

Are you running Bullseye?  Or something else?  

I'm going to try to reproduce on my end by installing the ADS1115 module, running for a bit, then switching to the Adafruit modules.  We'll see if I can run across the same issue.  I'm using Bullseye and PiFire v1.6.0-dev Build 40, so hopefully this will replicate it closely.  The only difference might be that I'm using a Pi Zero 2W.

nebhead, 2023-11-06:
Actually, can you send a link to your settings.json again?  That might shed some more light on the situation.

EricPalmquist, 2023-11-06:
[PiFire_11-06-23_164058.json](https://github.com/nebhead/PiFire/files/13273596/PiFire_11-06-23_164058.json)

nebhead, 2023-11-06:
> [PiFire_11-06-23_164058.json](https://github.com/nebhead/PiFire/files/13273596/PiFire_11-06-23_164058.json)

Oops, looks like I might have led you astray.  You have the ADS1015 device installed, not the ADS1115 device.  

```json
      "probe_devices": [
        {
          "config": {
            "ADC0_rd": "1000",
            "ADC1_rd": "1000",
            "ADC2_rd": "1000",
            "ADC3_rd": "1000",
            "i2c_bus_addr": "0x48",
            "voltage_ref": "3.2"
          },
          "device": "ADS1015Adafruit",
          "module": "ads1015_adafruit",
          "ports": [
            "ADC0",
            "ADC1",
            "ADC2",
            "ADC3"
          ]
        }
      ],
```

Can you run through the config wizard one more time and select the ADS1**1**15 Adafruit module instead?  Hopefully this resolves that issue.

EricPalmquist, 2023-11-06:
I'll go back to that.  I've been trying them all hoping to get lucky.  I'll go back to that one and try again.  I'm not a Python guru, but I think that I should not be getting the error shown below.  Kind of seems like my installation is corrupt somehow, but I've uninstalled it and re-installed a few times.

>>> from adafruit_platformdetect import Detector
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'Detector' from 'adafruit_platformdetect' (/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/__init__.py)
>>

nebhead, 2023-11-07:
Absolutely correct, shouldn't be seeing that error and would agree that it seems like corruption of some kind.  Maybe there are some python caches that need clearing somehow.  Let me look into that.

EricPalmquist, 2023-11-07:
I think I'm on to someting now.  I used the command below to force it to re-install and now the python command "from adafruit_platformdetect import Detector" works!  I'm going to fire pifire back up and see what happens.

 sudo python -m pip install --upgrade --force-reinstall adafruit-circuitpython-ads1x15

EricPalmquist, 2023-11-07:
Everything is working great now configured to use the "ADS1115 Adafruit" module instead of the "ADS1115".  My other issues had to do with Python libraries getting corrupted which is a seperate issue.  Since the documentation in the config Wizard states the Adafruit version is preferred I am going to close this issue.

closed 2023-11-07 by EricPalmquist