nebhead/PiFire #172: Unable to get the frontend to work

woudanator, 2023-12-21:
**To Reproduce**
Steps to reproduce the behavior:
1. get error 502 after install when trying to access the webui
2. I use a Rasberry PI zero v1.1
3. tried install on both rasbian bullseye and bookworm no luck

I am not sure if its something i am doing wrong or what is happning 

the supervisor logs show the below errors in the webapp 

[Traceback (most recent call last):
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/util.py", line 111, in load_class
    mod = importlib.import_module('.'.join(components))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/workers/geventlet.py", line 15, in <module>
    if parse_version(eventlet.__version__) < parse_version('0.24.1'):
AttributeError: module 'eventlet' has no attribute '__version__'
]

Error: class uri 'eventlet' invalid or not found: 

[Traceback (most recent call last):
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/util.py", line 111, in load_class
    mod = importlib.import_module('.'.join(components))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/workers/geventlet.py", line 15, in <module>
    if parse_version(eventlet.__version__) < parse_version('0.24.1'):
AttributeError: module 'eventlet' has no attribute '__version__'
]

Error: class uri 'eventlet' invalid or not found: 

[Traceback (most recent call last):
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/util.py", line 111, in load_class
    mod = importlib.import_module('.'.join(components))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/local/bin/pifire/lib/python3.9/site-packages/gunicorn/workers/geventlet.py", line 15, in <module>
    if parse_version(eventlet.__version__) < parse_version('0.24.1'):
AttributeError: module 'eventlet' has no attribute '__version__'

woudanator, 2023-12-21:
Just some more background 

Using a rasberry pi zero w v1.1 
I have no components connected at moment

nebhead, 2023-12-21:
Interesting.  Looks like a problem with eventlet's version.  I wonder if something went wonky during the installation process.  

Can you try SSH'ing into your system and attempting to update eventlet?  

At the command prompt: 
```
cd /usr/local/bin/pifire
source bin/activate
pip install --update eventlet 
```

If this doesn't work, then maybe grab your installed package list so we can check all the versions.
```
cd /usr/local/bin/pifire
source bin/activate
pip list 
```

nebhead, 2023-12-21:
Nevermind... I reproduced on my end.  Eventlet must have changed something on their end? <SMH> Anyway, stay tuned for a workaround.

nebhead, 2023-12-22:
The below should fix the issue.  This basically rolls back the eventlet module to a known working version.  I'm going to update the installer such that it fixes this issue.   Thanks for filing the sighting and sorry you ran into this. 

```
cd /usr/local/bin/pifire
source bin/activate
pip install eventlet==0.30.2 
```

woudanator, 2023-12-22:
Hi nebhead 

Not your faut thanks very much for the fix worked like a charm

nebhead, 2023-12-22:
> Hi nebhead
> 
> Not your faut thanks very much for the fix worked like a charm

You're very welcome!

closed 2023-12-22 by nebhead