nebhead/PiFire #177: PiFire error when changing modes too quickly

EricPalmquist, 2024-01-03:
Switching in and out of modes in less than a minute can cause an exception when the cookfile is created because the new name will be the same as the old name.

My suggestion is to include the current seconds in the file naming, meaning change line 74 from 
`	nowstring = now.strftime('%Y-%m-%d--%H%M')` to
`	nowstring = now.strftime('%Y-%m-%d--%H%M%S')`

https://github.com/nebhead/PiFire/blob/fc8c96ce44808b878b0fe477eed7baac2dd1c15f/file_mgmt/cookfile.py#L74C1-L74C44

nebhead, 2024-01-03:
Odd... this should only run after the Stop mode is engaged (and only if it wasn't in prime mode prior).  Can you explain more what reproduction steps you take and the error you are seeing?

EricPalmquist, 2024-01-04:
Quickly toggling between stop and run mode a few times does it.  I've only seen this running in the prototype mode on my PC in the debugger.  Probably unlikely that this would occur on an actual grill.

```
Exception has occurred: FileExistsError
[WinError 183] Cannot create a file when that file already exists: './history/2024-01-04--0738-CookFile'
  File "C:\Users\HP\OneDrive\Documents\piFire\mqtt-development\file_mgmt\cookfile.py", line 108, in create_cookfile
    os.mkdir(f'{HISTORY_FOLDER}{title}')  # Make temporary folder for all files
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\HP\OneDrive\Documents\piFire\mqtt-development\control.py", line 1130, in <module>
    create_cookfile()
FileExistsError: [WinError 183] Cannot create a file when that file already exists: './history/2024-01-04--0738-CookFile'
```

nebhead, 2024-01-04:
Ohhh, got it.  Sorry, my brain just wasn't calculating how this could happen but it's very simple.  LOL.  OK, I think we can make the change to add seconds into the mix or check if there is already another file there before writing a new file.  I'll take a look at it later today.  Thanks!

nebhead, 2024-01-04:
OK, fixed and pushed to the development branch.  Thanks for the bug sighting!

closed 2024-04-24 by nebhead