nebhead/PiFire #189: Control script startup fails with standalone controller in v1.8.0-dev11

ghandye, 2024-05-19:
**Describe the bug**
Control script does not start properly on current version of PiFire dev branch (v1.8.0-dev11) with my configuration.

Traceback shows from `grill_platform.get_input_status()` in `grillplat/raspberry_pi_all.py`.

Snippet from `control.err.log`:
```
INFO:control:Control Script Starting Up.
Traceback (most recent call last):
  File "/usr/local/bin/pifire/control.py", line 1058, in <module>
    last = grill_platform.get_input_status()
  File "/usr/local/bin/pifire/grillplat/raspberry_pi_all.py", line 153, in get_input_status
    return self.selector.is_active
AttributeError: 'NoneType' object has no attribute 'is_active'
```

Steps to reproduce the behavior:
1. Configure grill platform as a standalone controller
2. Try to start PiFire

**Expected behavior**
Normal startup of control script

**PiFire Version:**
 - 1.8.0-dev11

**Additional context**
Add any other context about the problem here.

ghandye, 2024-05-19:
I did a little more sleuthing on this...sharing in case it helps.

Trying to follow the logic, it looks like the code[1] in `grillplat/raspberry_pi_all.py` for `get_input_status()` checks if the grill platform is standalone or not here: 

```
                if not self.standalone:
                        self.selector = Button(self.in_pins['selector'])
                else:
                        self.selector = None
```
[1] https://github.com/nebhead/PiFire/blob/ede00b55/grillplat/raspberry_pi_all.py#L63-66

so in my case (standalone controller) that logic should hit the else statement and set `self.selector = None`.

When we get down to here[2] in the code:

```
        def get_input_status(self):
                if self.in_pins['selector'] is not None or self.standalone == False:
                        return self.selector.is_active
                return False
```
[2] https://github.com/nebhead/PiFire/blob/ede00b55/grillplat/raspberry_pi_all.py#L151-154

If I understand correctly, a standalone controller should fail that `if` statement since selector /is/ `None`, and it should just return `False`. But that doesn't seem to be happening, triggering the traceback and bug.

ghandye, 2024-05-19:
FYI, since I know my setup is standalone, I was able to hack around this on my controller by commenting out the `if` statement in L152-153 so it always returns `False`.

nebhead, 2024-06-10:
Thanks for the bug report and sorry it took me until now to look into this.  

Also, thanks for debugging it for me!   I changed the 'or' to an 'and' - so I'm hoping that will fix the issue.  Pushed to the development branch:  8f9488b793fddd23daceb6733827c21cb3f4bfb7 

Hopefully it works for you!

closed 2024-09-13 by nebhead