nebhead/PiFire #190: In-place upgrade from 1.7.0-dev54 to 1.8.0-dev11 fails to load grill platform module

ghandye, 2024-05-19:
**Describe the bug**
In-place upgrade from 1.7.0-dev54 to 1.8.0-dev11 (in development branch) fails to load the grill platform module.

`ERROR: An error occurred loading the [pifire_pwm] platform module.`

**To Reproduce**
Steps to reproduce the behavior:
Perform an in-place upgrade from 1.7.0-dev54 to 1.8.0-dev11

**Expected behavior**
Clean in-place upgrade.

**PiFire Version:**
 - 1.8.0-dev11

ghandye, 2024-05-19:
I was able to fix this particular error by simply running the config wizard again and selecting the proper platform. It looks like the new grill platform consolidation just needs some smoothing over for users migrating from 1.7.0.

nebhead, 2024-06-10:
Thanks again for the bug and the debug work too... let me try to fix this one up too.

nebhead, 2024-06-10:
OK, the upgrade path should now set the right grillplatform type in the settings modules.  Fixed in the last commit: 78b72abf6939110a4a351a78b187b010aeb20c6d

nebhead, 2024-09-13:
v1.8.0 is out today!  Hopefully the upgrade goes well.  If not, feel free to open a new issue.

closed 2024-09-13 by nebhead