nebhead/PiFire #192: Hopper Level is reported incorrectly on startup

maihde, 2024-05-31:
During startup, the current code in `control.py` calls `dist_device.get_level()` (https://github.com/nebhead/PiFire/blob/df50620f3abca0978e5f35cd103ac85fca090c2f/control.py#L214) without setting `override=True`.

This often means the initial reported pellet level will be reported as 100% until the next pellet measurement is taken during the regular loop.

Additionally, I noted that the `vl53l0x` module implementation of `get_level` when `override=True` uses a simple `sleep` to wait for the sensing loop to complete.  While this will generally work, it's probably better to convert to using `threading.Event` as it would allow the function to return quicker once the read is completed, thus permitting a longer timeout as well.  This is somewhat of an feature/optimization so it doesn't need a fix, but I'm happy to contribute a pull-request.

**Expected behavior**
The initial hopper level is reported correctly on startup.

**Suggested fix**
In the inital call to `get_level` during startup set `override=True`

**PiFire Version:**
 - Version [e.g. 1.7.1]

nebhead, 2024-05-31:
Excellent feedback!  I'll incorporate this fix into v1.8.0.

nebhead, 2024-09-13:
I believe this should be resolved in v1.8.0, but feel free to re-open if you still see a bug.

closed 2024-09-13 by nebhead