nebhead/PiFire #195: Add support for BLE meat thermometers for less soldering

bobpaul, 2024-06-12:
This was described in a [comment](https://github.com/nebhead/PiFire/issues/149#issuecomment-1435932324) on #149, and I thought it would be a good idea to add it here so it's not forgotten:

Basically the idea is use a cheap BLE meat thermometer with several inputs ( @faisalhusain007 mentioned one with 6 probes for $20 ) as meat thermometers to PiFire. Someone could use this in a build instead of using the onboard ADC. PiFire would then make the values available via its webapp and use them for cooking modes that utilize the meat probes.

nebhead, 2024-06-12:
This is definitely still in the queue.  I had recently come up with the concept of 'transient' probes which is more-or-less implemented in the latest development branch.  This would allow a probe _device_ to be transient, meaning not always there which might be the case with a bluetooth device.  That said, I haven't yet purchased any of these devices nor have I invested time to dig into the code to support them yet.  In summary... it's coming someday, but don't hold your breath for it to happen soon.

palmtreefrb, 2024-12-19:
For the "Meater" bluetooth probes there is a python library here. It works quite well. [https://github.com/nathanfaber/meaterble](https://github.com/nathanfaber/meaterble)

nebhead, 2024-12-19:
All - This is now implemented on the development branch (v1.9.0).  Just pushed the Meater Pro support yesterday.  It's still very experimental, but if you have any Meaters (I do not, so I am not really able to test), please give it a spin and let me know.

nebhead, 2025-04-18:
Closing with the recent addition of BLE probes to v1.9.0 official release.

closed 2025-04-18 by nebhead