nebhead/PiFire #198: Relays , active low not working. V2.00a pcb

retroscl, 2024-09-14:
Relay issue with v1.8.0 using pi zero 2W - on v2.00a PCB. Active low relay option does not appear to work

**To Reproduce**
Steps to reproduce the behavior:
Run wizard, select active low relay option. Drives a 4 relay active low module

**Expected behavior**
After wizard restart, all relay gpio's are at zero volts. Relays respond in reverse. Head into manual mode, enable F,A,I,Power relays - check gpio's and all are at 3.28V.
Tried setting active hi in wizard and it gives the exact same logic levels which is what I expected for logic hi. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**PiFire Version:**
 - Version [e.g. 1.8.0]

**Additional context**
Add any other context about the problem here.

nebhead, 2024-09-14:
Thank you for the bug report!  I think I fixed the issue with the latest push.   https://github.com/nebhead/PiFire/commit/295ad672d15f53de236a56756d457da01b4172f5  

Do an another update and this should be resolved.  Let me know.

retroscl, 2024-09-14:
Perfect,  active low relay logic fixed. Will continue with build.
Thanks

closed 2024-10-14 by nebhead