nebhead/PiFire #199: Pellet-Manager calculate pallets wrong

benixs, 2024-09-29:
On the Pellet-Manager it displays a user over a tonne of Pellets. 
I have a feed Rate of 0.9g/s maybe it calculate it wrong. Then  Prime ist on 20grams and it calculate 361kg...

Prime in  Data 22s and 20 gramm

[image]

**PiFire Version:**
 - Version 1.8.1

nebhead, 2024-09-30:
That sure looks like a bug!  I think it's actually the time that is calculated incorrectly in the metrics log.  I'll look into this and hopefully get this one squashed.  Thanks for the issue submission!

nebhead, 2024-09-30:
Did you by any chance adjust the time/date during the run?  

Start time was: 
[image]

End time was: 
[image]

nebhead, 2024-10-14:
I'm going to close this.  Feel free to open again if you want to continue to debug.

closed 2024-10-14 by nebhead