nebhead/PiFire #200: Update from 1.7.1 to 1.8.3 caused scikit-learn missing dependency

maihde, 2024-10-13:
**Describe the bug**
After updating (via the webapp) from 1.7.1 to 1.8.3 this error message was preventing the controller from launching.  The webpage was complaining about a possible redis issue, but investigation of the logs indicated it was a missing dependency.

```
Traceback (most recent call last):
  File "/usr/local/bin/pifire/control.py", line 28, in <module>
    from notify.notifications import *
  File "/usr/local/bin/pifire/notify/notifications.py", line 27, in <module>
    from sklearn.linear_model import LinearRegression
ModuleNotFoundError: No module named 'sklearn'
```

The workaround was to log into PiFire via ssh and run

```
. /usr/local/bin/pifire/bin/activate
pip install scikit-learn==1.4.2
```

nebhead, 2024-10-14:
Thanks for submitting this issue!  I'll look into this soon and hopefully get it fixed up.

nebhead, 2024-10-14:
I think I have this fixed in the upgrade path now.  Thanks again for pointing it out.  I'll push an update soon.

closed 2024-10-14 by nebhead