nebhead/PiFire #201: Modify the "An error occurred reading the voltage from one of the ports" to be more helpful.

maihde, 2024-10-13:
**Is your feature request related to a problem? Please describe.**

When a temperature probe input is left unconnected I am continually receiving this message in the logs:

```
... 489 additional messages suppressed
An error occurred reading the voltage from one of the ports. The voltage read (3309.375mV) was outside the expected range of 0mV to 3200.0mV
```

The frequency of this message is causing a lot of messages to be suppressed.

**Describe the solution you'd like**

1) Include the port that is failing in the error message.  This will make it easier for a user to identify the source of issue.
2) Consider changing the error message to a debug level, since disconnected ports are a common normal operation.

nebhead, 2024-10-14:
Thanks for submitting this issue!  I'll look into this soon and hopefully get it fixed up.

nebhead, 2024-10-14:
I'll modify the error message to include the port and more useful information about a fix.  I'll push an update soon.

nebhead, 2024-10-14:
BTW - You should consider changing your voltage reference to 3.35V in the ADC device in your configuration wizard.  This will improve your temperature calculation accuracy, as your actual voltage reference appears to be ~3.3V.   

Pushing updates soon.

closed 2024-10-14 by nebhead

maihde, 2024-10-16:
@nebhead agreed about the updated my voltage reference!

In my specific case, I have the 3.01a PCB and I'm not using the 4th probe input.  Would there be any value in having the option in PiFire use this extra ADC input to measure `Vs` either once at boot or in real-time so that other probes could adjust automatically?  If so, I would be happy to implement this feature and submit a pull request.

nebhead, 2024-10-16:
> In my specific case, I have the 3.01a PCB and I'm not using the 4th probe input. Would there be any value in having the option in PiFire use this extra ADC input to measure `Vs` either once at boot [...]

Interesting idea!  If you want to take a crack at that, sure!  If you do, please make sure you do this on the development branch and grab the latest before raising the PR.