nebhead/PiFire #202: Control process script expericed a timeout.  Webapp has no error notifcation and temperature values displayed are stale

maihde, 2024-10-13:
**Describe the bug**
During a cook, PiFire stopped responding but the webapp continued to display valid temperature values and did not show an error.  When looking at the logs, this message was present:

```
The control process experienced a timeout event (no heartbeat detected in 30 seconds) and is being reset.
```

Although the log says the control process was being reset, it did not appear to reset and the webapp continued to show stale temperature values (making it difficult to recongize that this condition occured).

Other logs aren't timestamped, but here are some potentially concerning lines:

In control.out.log:
```
error: <class 'PermissionError'>, [Errno 13] Permission denied: file: /usr/lib/python3/dist-packages/supervisor/xmlrpc.py line: 557
```

In control.err.log:
```
TabError: inconsistent use of tabs and spaces in indentation
  File "/usr/local/bin/pifire/control.py", line 1093
    controlLogger.info(f"OUTPUTS {settings['outpins']}")
```

**To Reproduce**
Unknown.  This is the only time it has happend.

**Expected behavior**
In the event the control processe hangs or doesn't restart, the webapp should ideally show and error notification and set the temperature values to zero.

**PiFire Version:**
 - Version 1.7.1

nebhead, 2024-10-14:
Thanks for submitting this issue!  I'll look into this soon and hopefully get it fixed up.

nebhead, 2024-10-14:
I believe this line of code is removed in the latest release, as I can't find it.  Also, there is new code to check the health of the control script and report if the script is unresponsive on the dashboard.  Let me know if you face this one again.  For now, I'm going to close this issue but feel free to reopen if you see it on v1.8.x.

closed 2024-10-14 by nebhead

maihde, 2024-11-25:
This happened again today @nebhead .  Here are some error log messages.

```
Exception in thread Thread-7 (_display_loop):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/bin/pifire/display/ssd1306.py", line 105, in _display_loop
    self._display_current(self.in_data, self.status_data)
  File "/usr/local/bin/pifire/display/ssd1306.py", line 142, in _display_current
    with canvas(self.device) as draw:
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/render.py", line 43, in __exit__
    self.device.display(self.image)
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/oled/device/__init__.py", line 311, in display
    self.data(list(buf))
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/device.py", line 55, in data
    self._serial_interface.data(data)
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/interface/serial.py", line 128, in data
    write(list(data[i:i + block_size]))
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/interface/serial.py", line 137, in _write_large_block
    self._bus.i2c_rdwr(self._i2c_msg_write(self._addr, [self._data_mode] + data))
  File "/usr/lib/python3/dist-packages/smbus2/smbus2.py", line 658, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 5] Input/output error
Traceback (most recent call last):
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/redis/_parsers/socket.py", line 65, in _read_from_socket
    data = self._sock.recv(socket_read_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TimeoutError: [Errno 110] Connection timed out
```

```
Exception in thread Thread-7 (_display_loop):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/bin/pifire/display/ssd1306.py", line 105, in _display_loop
    self._display_current(self.in_data, self.status_data)
  File "/usr/local/bin/pifire/display/ssd1306.py", line 142, in _display_current
    with canvas(self.device) as draw:
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/render.py", line 43, in __exit__
    self.device.display(self.image)
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/oled/device/__init__.py", line 311, in display
    self.data(list(buf))
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/device.py", line 55, in data
    self._serial_interface.data(data)
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/interface/serial.py", line 128, in data
    write(list(data[i:i + block_size]))
  File "/usr/local/bin/pifire/lib/python3.11/site-packages/luma/core/interface/serial.py", line 137, in _write_large_block
    self._bus.i2c_rdwr(self._i2c_msg_write(self._addr, [self._data_mode] + data))
  File "/usr/lib/python3/dist-packages/smbus2/smbus2.py", line 658, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 5] Input/output error
```

nebhead, 2024-11-25:
This looks like a different issue with interfacing the I2C within a thread.  Could be an issue with the display being unresponsive.  At least lit looks like in both cases above it's trying to display something to the OLED screen.  Does this happen frequently?  We could try to reduce the I2C bus frequency.

maihde, 2024-11-28:
@nebhead - it's happened a couple times now.  What do you think about ensuring that the _display_loop is surrounded in a try/except so that unexpected errors don't cause the loop to exit?

```
while True:
    try:
        ... existing code ...
    except:
        logging.exception("unexpected error in display loop")
```

nebhead, 2024-11-28:
That could work, though the I2C may be dead at that point and may fail again.  It's worth a shot though.  Feel free to give it a go.  Did you try to reduce the I2C bus speed?  We could also increase the loop delay to see if that helps:

`time.sleep(0.1)` to `time.sleep(0.2)`