nebhead/PiFire #204: 1Wire pin not selectable within configuration

ew-design, 2024-11-15:
**Describe the bug**
The drop down in the configuration for the 1-wire setup doesn't include GPIO4 which is tied to the 3pin JST connector on the newest board. 

**To Reproduce**
If you go to config it only offers other GPIOs

**Expected behavior**
The config should allow any pin that is not currently assigned to the relays etc to be used for 1-wire.

nebhead, 2024-11-15:
Did you select the custom configuration?  

The PCB 4.x.x configuration _only_ has 1wire at GPIO4 and is hidden since there isn't an option to change it.  

[image]

nebhead, 2024-11-16:
Looks like the custom configuration is missing the GPIO 4 option.  I'll update that and push the change out.  Thanks for the issue submission.

ew-design, 2024-11-16:
Thank you for your extremely quick response. I was running through my
probes trying to calibrate and I have like 3 of those 1-wire ds18b20s lying
around.

I didn't even have a chance to login to my pifire since your first reply
and you're already pushing a fix. Thank you so much! Have a great weekend
dude!

nebhead, 2024-11-16:
v1.8.8 Pushed and ready to download.  Enjoy.

closed 2024-11-16 by nebhead

ew-design, 2024-11-16:
Awesome, thanks!