nebhead/PiFire #207: Problem after SW update and re-install

thomasstrach, 2024-11-18:
Hi, 

as performed several times before I used the update function to get the latest version on my RPi controller.
Unfortunately it ended up with 2 error messages:
"Unable to reach Redis database. You may need to reinstall PiFire or enable rdis-server.
The control process did not respond to a request and may be stopped.

So I reinstalled PiFire with: 
curl https://raw.githubusercontent.com/nebhead/pifire/main/auto-install/install.sh | bash

and ended up in a different error on supervisor page:
Webapp fatal error.
can't find command '/usr/local/bin/pifire/bin/gunicorn'
[image]

and this error log message:

n /usr/local/lib/python3.9/dist-packages (from adafruit-circuitpython-typing->Adafruit-Blinka->adafruit-circuitpython-ads1x15) (1.14.0)
Return Code: 0
Reading package lists...
Building dependency tree...
Reading state information...
python3-smbus is already the newest version (4.2-1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Return Code: 0
command output: PiFire Board Configuration Tool v1.0.0
command output: Ben Parmeter - 2024 - MIT License
command output: --help, -h for command details
command output: 
command output: Results:
command output: - Setting the 1Wire pin: SUCCESS
command output: - Enabling SPI: SUCCESS
command output: - Enabling I2C: SUCCESS
command output: - Attempting to append data to /etc/modules:  SUCCESS (appending file /etc/modules)
command output: PiFire Board Configuration Tool v1.0.0
command output: Ben Parmeter - 2024 - MIT License
command output: --help, -h for command details
command output: 
command output: Results:
command output: - Enabling SPI: SUCCESS

nebhead, 2024-11-18:
It sounds from your description that at least the second time you installed pifire, it didn't install correctly.  Did you wipe the SDCard before running the installation script?

thomasstrach, 2024-11-25:
Hi,
Reinstalling was not that easy as the raspberry did not boot with network connectivity. Nevertheless after new image and re-installing latest version the error messages are gone but for whatever reason the auger does not work anymore. All other Relais are working fine, only the auger output is not providing any signal. 
Do you have any idea? 
Even in manual mode I can see the icon on the display but the relay gets no signal.
Any special config at raspberry needed?

thomasstrach, 2024-11-25:
I am running HW board ver 2.00 with a raspberry zero 2W, mechanical Relais Modul, SPI Display, Encoder Button and vl53l0x sensor. Do you need any log file?

nebhead, 2024-11-25:
Are you now on the latest PiFire (main branch)?  v1.8.x? 

In the configuration wizard, are you selecting the 2.00a PCB? 

[image]

Can get the settings.json file (let me know if you need help getting this) and post the platform configuration section?   It should look something like this (yours will have different values):

```json
  "platform": {
    "buttonslevel": "HIGH",
    "current": "custom",
    "dc_fan": true,
    "devices": {
      "display": {
        "dc": 24,
        "led": 5,
        "rst": 25
      },
      "distance": {
        "echo": 27,
        "trig": 23
      },
      "input": {
        "down_dt": 20,
        "enter_sw": 21,
        "up_clk": 16
      }
    },
    "inputs": {
      "selector": 17,
      "shutdown": 17
    },
    "outputs": {
      "auger": 14,
      "dc_fan": 26,
      "fan": 15,
      "igniter": 18,
      "power": 4,
      "pwm": 13
    },
    "real_hw": false,
    "standalone": true,
    "system": {
      "1WIRE": 6,
      "SPI0": {
        "CE0": 8,
        "CE1": 7
      }
    },
    "system_type": "prototype",
    "triggerlevel": "HIGH"
  }
```

thomasstrach, 2024-11-25:
Hi,

 

yes I am on latest version now:

System Update (Current Version: 1.8.8)

 

I have chosen the Version 2.00a PCB during config.

The other 3 relais are working fine on the board even when I switch the cables (auger vs. fan).

It looks like GPIO14 is not providing any output at all, neither high or low.

 

Settings.json is attached, 

 

But where do I find the platform configuration settings?

 

image.png (view on web) <https://github.com/user-attachments/assets/63b85e16-1315-480e-9039-3148020839e0>

nebhead, 2024-11-25:
>  Settings.json is attached, But where do I find the platform

Unfortunately the settings.json did not come across in GitHub.  If you look at the file, you should see a section that says: "platform".

thomasstrach, 2024-11-25:
Ah, now I got it…

 

"platform": {

    "buttonslevel": "HIGH",

    "current": "pcb_2.00a",

    "dc_fan": false,

    "devices": {

      "display": {

        "dc": 24,

        "led": 5,

        "rst": 25

      },

      "distance": {

        "echo": 27,

        "trig": 23

      },

      "input": {

        "down_dt": 20,

        "enter_sw": 21,

        "up_clk": 16

      }

    },

    "inputs": {

      "selector": 17,

      "shutdown": null

    },

    "outputs": {

      "auger": 14,

      "dc_fan": null,

      "fan": 15,

      "igniter": 18,

      "power": 4,

      "pwm": null

    },

    "real_hw": true,

[... 468 lines cut ...]
    "start_to_mode": {

      "after_startup_mode": "Smoke",

      "primary_setpoint": 165

    },

    "startup_exit_temp": -17

  },

  "versions": {

    "build": 36,

    "cookfile": "1.5.0",

    "recipe": "1.0.0",

    "server": "1.8.8"

  }

}

nebhead, 2024-11-25:
That all looks OK.  Is there a possible issue with your relay?   Or the wiring from the relay header?  Do you have a multimeter?  Can you test the relay header is outputting high or low when expected?

thomasstrach, 2024-11-26:
Well it is really strange. In the end I used the manual config with GPIO4 instead GPIO14 for the auger and changed the input cables at the relais card.

The power relais is not working now on GPIO14 but GPIO4 is turning the auger on.

Encoder switch is not working now, will have a look on this later on.

nebhead, 2024-11-26:
This pin is also used as a UART pin if configured, so hopefully that didn't get turned on be default or anything.  Normally this is not configured for UART.   You can check your /boot/firmware/config.txt to see if there is any lines that are enabling UART.

nebhead, 2024-11-26:
OK, that all looks fine.  Well, except for the PWM pin being activated.  I'd recommend adding a # in front of that line to ensure PWM is not enabled.  Assuming you are not using a DC Fan on this build.  

So when you measure voltage of GPIO14, are you changing the output with manual mode?  Does it still not switch?  That might indicate that the GPIO14 is damaged/not working.  What we might be able to do is to utilize another spare GPIO to the relay instead.  Basically tie another of the GPIO's from the AUX header that you are not currently using - we can set the settings.json appropriately to utilize the other GPIO.

thomasstrach, 2024-11-27:
First of all I am really mad about myself, why did I changed anything on a proper running system… ☹

 

Maybe it’s a better idea to downgrade to an older version?

 

Looking at the /boot/firmware/config.txt file, where to put the #?

Do you mean it like this?

 

enable_uart=0

dtparam=act_led_trigger=actpwr

# dtoverlay=pwm,func=4,pin=13  # Modified by PiFire Board Configuration Utility

 

Meanwhile the controller is working remotely with the manual setup and using GPIO4 for the auger instead of GPIO14 and it keeps the set temperature/mode fine.

 

I am using the ili9341 with the rotary encoder, this encoder is not working at all.

No matter which of the 2 encoders I am choosing during setup wizard.

 

The display is working after startup until I have changed the “set temp” or “workmode” several times. 

After doing this changes the display turns blank, just shining bright, the webinterface and the smoker itself continues running.

Once I noticed that with a blank display the notification for running out of pellets continues sending alarms despite I filled up the pellets and the webinterface shows 100%.

 

When I push the stop button on the web interface the system stops immediately without the 240 sec. fan cool down time. Is this behavior ok?

During startup I am missing the prime functionality before ignite function. 

Is this available by starting via the web interface?

thomasstrach, 2024-11-27:
BTW. I found the Prime and Start functionality on the web interface!

nebhead, 2024-11-27:
> First of all I am really mad about myself, why did I changed anything on a proper running system… ☹ Maybe it’s a better idea to downgrade to an older version? 

Don't beat yourself up.  We'll figure this out.  You can try to downgrade, but I'm willing to bet you're going to have issues.  

> Looking at the /boot/firmware/config.txt file, where to put the #? Do you mean it like this? 
> enable_uart=0 
> dtparam=act_led_trigger=actpwr 
> # dtoverlay=pwm,func=4,pin=13 # Modified by PiFire Board Configuration Utility 

Yep, perfect.  

Do you see any other lines in there, for example for 1-Wire that are enabled?   We definitely want to disable 1-Wire and have had reports of the new version turning it on by mistake. 

> Meanwhile the controller is working remotely with the manual setup and using GPIO4 for the auger instead of GPIO14 and it keeps the set temperature/mode fine. 

So, that's encouraging.  Still strange that the GPIO14 is misbehaving.  Do you have a different Pi by any chance?  We could not even hook it up to the board, and install PiFire, just to test if GPIO14 behaves differently.  

> I am using the ili9341 with the rotary encoder, this encoder is not working at all. No matter which of the 2 encoders I am choosing during setup wizard. The display is working after startup until I [...]

Can you take a screenshot of the one that you are using the configuration wizard?   It sounds like it might be a config issue that I've seen before.  Also, I've seen this with loose wiring to the display.  Unless this is consistent, there are a few things that can be touchy here.  

> Once I noticed that with a blank display the notification for running out of pellets continues sending alarms despite I filled up the pellets and the webinterface shows 100%. 

This might be an actual bug.  I'll look into this. 

> When I push the stop button on the web interface the system stops immediately without the 240 sec. fan cool down time. Is this behavior ok? 

Yes, the stop button will cause it to stop immediately where as the little finish flag will go through a shutdown cycle for 240s.  The finish flag only shows up after you exit startup mode.  

> During startup I am missing the prime functionality before ignite function. Is this available by starting via the web interface? 

Looks like you found it!

thomasstrach, 2024-11-28:
Thanks for your effort with my problems, really very much appreciated!

 

See my comments below:

 

Ok, I will continue with this version and hope you can help me to get it fixed.

Just implementent the change, will have to test it.

I cannot find anything in this file regarding 1-wire.

I have just ordered another raspberry  zero 2W to test this out. Will take some days…

I used GPIO 20/21/16 for the button Down/Enter/Buttom up config (pre-set), that’s where my encoder is connected.

I have tried out both selections for the encoder

 

Ah, I only used the shutdown on the display with the rotary encoder before.

Yes, I used the display with encoder before for startup.

nebhead, 2024-11-28:
> I have just ordered another raspberry  zero 2W to test this out. Will take some days…

This is great - it should help us to understand if what we are seeing is a hardware problem with the existing Pi.

thomasstrach, 2024-11-29:
Just received the controller and things have turned around.

After formatting the sd card, now the WIFI was connected automatically, that was not happening before with several attemps.

All is running fine, no problems anymore.

Display is working, encoder and all GPIOs.

I will try again when I have time with old Pi, maybe the SD card and/or the Pi is wrong.

Thanks a lot for your patience and support!

nebhead, 2024-11-29:
Excellent news!  You are very welcome.  Hope all is well from here on out.

closed 2024-11-29 by nebhead