nebhead/PiFire #208: Error with “Hold Temp” setting in operating mode with C°

Buetzu, 2025-01-05:
If during operation with Celsius (C°) the setting:

1. “Settings”
2. “Startup & Shutdown”
3. “Goto Mode After Startup” to “Hold Mode”
4. “Hold Temp” is set to 95°C

When saving the value, the error message “The value must be greater than or equal to 200” appears. 
it looks as if the value is not Celsius but 200°F (93.33°C) is expected as the minimum temperature. 

PiFire Version: 1.9.0
PiFire Build: 13
[image]

nebhead, 2025-03-04:
Apologies!  I just realized that this has been sitting around waiting for me to respond for two months!  Sorry about that. 

Anyway, this particular field is using the the **Maximum Startup Temp** from the Safety settings to set a minimum value here.  

```
min="{{ settings['safety']['maxstartuptemp'] }}" max="{{ settings['safety']['maxtemp'] }}"
``` 

I could un-restrict the minimum, but then you start bumping into potential issues.   There may be another bug here... but I'll tackle that separately.

Can you double check your config and see if your safety Max Startup Temp is set to this value?

Buetzu, 2025-03-04:
No problem, thanks for your feedback. I have now set the “Max Startup Temp (C)” value to 90°C in the “Safety” tab. Then in the tab “Startup & Shutdown” on “Hold Mode” with “Hold Temp” can be fixed to 90°C without error message. 

 Problem solved :-)

[image]

[image]

nebhead, 2025-03-04:
Wonderful!  Thank you!