nebhead/PiFire #210: ST7789 240x240 display just black display. [Fix included]

markalston, 2025-03-19:
My [ST7789](https://www.amazon.com/dp/B07MH93747)  would not function with the current pifire.  The display was simply black with nothing displaying.

Switching to the luma.lcd python module (by copying the ili9341e support) with some bus speed changes fixed the display and it now works perfectly.

I have a pull request that fixes the display and adds rotary encoder support.
#209

nebhead, 2025-03-19:
Oh awesome, thanks for the pull request.  I'll try to take a look at it and approve this week.  Things have been a bit busy at the day job. :)

markalston, 2025-03-20:
Added pull request to work against development branch #211

markalston, 2025-03-29:
resolved with pull #215

closed 2025-03-29 by markalston