nebhead/PiFire #234: Meater probe does not work

thomasstrach, 2025-07-21:
**Describe the bug**
No value for Meater probe in UI

**To Reproduce**
Steps to reproduce the behavior:
I have just updated to latest version. 1.9.4
When I use the installation wizard to install the Meater probe, all is running fine.
System is scanning and detecting the probe with two sensors that I can add to the UI.
Unfortunately after the reboot these 2 sensors show no value and no connectivity in monitor mode.

What am I doing wrong?

Thanks in advance

thomasstrach, 2025-07-21:
> Which Meater model do you have?
> What Probe Device Module are you using for Meater support?  The normal Meater module or the experimental module?

Hi,I have a brand new Meater plus and during install I used the normal Meater module.The scan finds the probe successfully.

nebhead, 2025-07-21:
OK, can you send some logs?  

Do you know if the Meater Plus has a block that also shows up as a BT device?  You may have to turn off or remove the battery from the block so that PiFire doesn't try to connect to that device instead.

nebhead, 2025-08-05:
Great to hear!  I'll go ahead and close the issue.